PaulStoffregen/Audio #106: AudioAnalogInput dc offset issues

jeff-hamm, 2015-02-10:
The current dc offset code struggles a bit in real world scenarios. 
It works more or less as expected if you're powering the board & teensy via USB, however, on a disconnected board that includes a DC offset WITH capacitors, or if there is an unbalanced AC signal on the line when the board first boots up, the DC offset will be incorrect.

Currently, there is no way in the library to insert a delay between construction of the AnalogInput object and the initial reading of the DC offset to allow the capacitors to charge up (or to let other noise settle).

Furthermore, in the update function, there is a rolling DC offset adjustment function that doesn't typically solved the problem:
		tmp = (uint16_t)(*p) - (int32_t)dc;
		s = signed_saturate_rshift(tmp, 16, 0);
		*p++ = s;
		dc += s / 12000;  // slow response, remove DC component

this increments the dc offset only if it is greater than 12000 from the current offset. This prevents it from getting closer than 5.4% to the true offset, which, when talking about mV audio signals is pretty significant, so, most of the time we're stuck with the dc offset we booted up with.

I'd make either change, however the first requires an architectural change that I'm not sure the ideal solution for (locally I solved the problem by removing 'init' from the constructor and calling it deliberately when I'm ready). And the dc offset limitation is clearly intentional, so I'd need to know more about the intention before submitting a change.

Thanks for the wonderful library!

kpckpc, 2015-04-18:
Just signed up to github, and noticed this message.

As long as a signal with  an amplitude > 12000/2, the dc will actually settle, the signal will act like dithering. The problem you describe will only occur when no signal is present.

The following (untested) code should not have problems with truncation. Futhermore it replaces a divide by two multiplies, which should be faster.

        // filter coefficients, slow response, remove DC component
        const uint32_t alpha = 4294967296. / 12000.;
        const uint32_t beta = 4294967296ull - alpha;
        do {
                uint32_t sample = (uint16_t)(*p);
                dc = ((uint64_t) alpha * sample + (uint64_t) beta * dc) >> 32;
                s = signed_saturate_rshift(sample - dc, 16, 0);
                *p++ = s;
        } while (p < end);

One other thing is that this calculates the actual DC offset of the input, whereas the Audio library version calculates it on the saturated signal. But if you start saturating, the DC level was ruined anyway, before or after should not matter.

PaulStoffregen, 2016-12-10:
(hopefully) fixed by https://github.com/PaulStoffregen/Audio/commit/dde07982b06ef94fa1b9a3a07d16867e1b653e79

closed 2016-12-10 by PaulStoffregen