PaulStoffregen/Audio #111: Glitches at zero-crossings of waveforms on DAC?

endolith, 2015-03-27:
Not sure if this is an off-by-one bug or hardware imperfection or what, but I see positive spikes at the zero-crossings of synth waveforms when running through the DAC, but not when running through the CODEC board. Simple example:

```
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>

AudioSynthWaveform   waveform;
AudioOutputAnalog    audioOutput;
AudioConnection patchCord3(waveform, 0, audioOutput, 0);

void setup() {
  AudioMemory(12);
  waveform.begin(0.1, 1000, WAVEFORM_TRIANGLE);  // wave with glitches
//  waveform.begin(0.001, 1000, WAVEFORM_TRIANGLE);  // shows only glitches
//  waveform.begin(0, 1000, WAVEFORM_TRIANGLE);  // no glitches
}

void loop() {
}
```

At full amplitude they're not visible:

[image]

but at 0.1 they are:
[image]
[image]
[image]

[image]

with external CODEC I don't see any, even at 0.1 amplitude:
[image]

Nantonos, 2015-03-27:
Nicely done report. So you are seeing these on two waveforms (ramp and triangle) and on both rising and falling slopes. most severe at zero crossing but also lesser spikes elsewhere. The external codec is the SGTL5000 audio board?

As you have no reconstruction filter, I wonder if this is aliasing, or sampling frequency breakthrough. Does varying the frequency of the waveform change the glitch pattern at all?

This could be a MSB DAC glitch though. The datasheet is not as detailed here as one would expect for an external DAC chip, for example.

endolith, 2015-03-27:
It was on sine also, I'm not sure about square, I didn't look closely at the zero crossings.

> The external codec is the SGTL5000 audio board?

Yes, [this board](https://www.pjrc.com/store/teensy3_audio.html)

> Does varying the frequency of the waveform change the glitch pattern at all?

No, it was always around zero crossings, no matter what frequency.  I guess I didn't check extremely low or high frequencies, though, only around 1-4 kHz.  Also I couldn't figure out which frequency would make the spikes a stable distance apart; period would vary by ±1 sample

PaulStoffregen, 2015-03-27:
Yup, running the code and I see it here too:

[image]

PaulStoffregen, 2015-03-27:
I'm afraid this may simply be a hardware limitation that requires external analog filtering.  Maybe?

Here's a zoom in to just 1 spike.  It's barely 1 us wide, much less than the 22.7 us period where a new value is being written to the DAC register.

[image]

endolith, 2015-03-28:
They're still present with the scope's digital filter at 20 kHz, though, so part of it is in the audio band already:

[image]

with the scope's "Peak Detect" mode, they show up on every cycle, and there are smaller ones at different amplitudes that are the same every cycle:

1 Hz, 0.1 amplitude:
[image]

[image]

1 Hz, 1 amplitude:
[image]

Nantonos, 2015-03-28:
Yes, this looks like a "major carry transition glitch".
http://e2e.ti.com/blogs_/b/analogwire/archive/2013/06/14/what-s-with-all-this-glitch-ing
http://www.maximintegrated.com/en/app-notes/index.mvp/id/583
http://digital.ni.com/public.nsf/allkb/9B13E25B0B0EC197862562B4007D2E33

There is also a lot of high-frequency hash on top of the waveform. The K20P64M72SF1 datasheet is entirely silent on the matter of glitch energy. It does suggest a 47pF cap at the output.

Comparison of the triangle produced by the sketch at the head of this post (1kHz triangle) with that produced by an analog oscillator (Intellijel Dixie II):
[image]
[image]

endolith, 2015-03-29:
yes, that looks like it.  so it's not a bug, just a hardware limitation.

closed 2015-05-01 by endolith