PaulStoffregen/Audio #129: Import function

mamuesp, 2015-12-08:
For the moment, the import function is not very error tolerant. So it is necessary to enter the code between the to comment lines:
_// GUItool: begin automatically generated code_
_// GUItool: end automatically generated code_
and the lines must have a least two "words" separated by a whitespace - if there is no whitespace, the pattern matching regex will crash.
This will be fixed in a future version.

PaulStoffregen, 2015-12-08:
Any chance you might take a look at issue #59.  Is this sort of things practical to do?

Especially as the library grows with more types of hardware objects, conflicts for the fixed hardware resources are going to be come a really confusing problem.

mamuesp, 2015-12-08:
I believe this is not a big problem - the extension with the link to the hardware parameters and/or pins to read or write was one goal when I started looking into the code. So after I made my importer more tolerant (you should be able to throw "Moby Dick" in it without any error coming up ...) ;-), I will look into the mentioned issue.

My little goal is to be able to assemble a kind of "minimoog" with the tool - if that is possible, I believe also others would like to play along ...

closed 2015-12-26 by PaulStoffregen