PaulStoffregen/Audio #140: GUI documentation URLs are broken

PaulStoffregen, 2015-12-13:
Recent changes seem to have broken the URL updating to append the currently selected documentation.

Here's the original commit which added this feature: https://github.com/PaulStoffregen/Audio/commit/029bbe268f5ec26d7d3cb88a2421ef601f5a053f

This URL update allows people to copy the URL from the address bar into a forum or email message, when they wish to answer a question with a link to specific documentation.

mamuesp, 2015-12-13:
> Here's the original commit which added this feature: 029bbe2 <https://github.com/PaulStoffregen/Audio/commit/029bbe268f5ec26d7d3cb88a2421ef601f5a053f>

The function selectNode carries the lines which are responsible for this functionality. They had be turned into a comment in the version which is base of my fork. So it was set off already by yourself before the fork. I reactivated it now, with the next commit, all will be fine.

mamuesp, 2015-12-13:
Done.

mamuesp, 2015-12-14:
I found the reason why these two lines had been commented out: http://bit.ly/1NZ5Zuu

closed 2015-12-26 by PaulStoffregen