PaulStoffregen/Audio #144: More waveforms for LFO purposes

floretan, 2015-12-27:
The current synth waveforms also work as LFOs, but it would be nice to have a few extras, like reverse sawtooth wave or a sample-and-hold. Since these only make sense as LFOs, it's probably not appropriate to add them to the synth waveform object, should they be created as a separate object? Ideally I'd like to contribute this back to the audio library, so whatever input on the direction makes that more likely to happen later on would be helpful.

PaulStoffregen, 2015-12-27:
These should probably be done as part of the existing waveform object.

floretan, 2015-12-28:
Thanks for the quick feedback, it turned out to be a lot easier than I thought (see pull request). The only tricky part was to figure out when a new random sample needs to be done for the "sample and hold" waveform. This works fine, but maybe there's a more elegant way to do it than checking ``tone_phase < tone_incr``.

PR #145 (2015-12-28): More waveforms
The reverse sawtooth and sample-and-hold waveforms are quite practical as LFOs. This change resolves #144, more waveforms can be added later if needed.
closed 2015-12-28 by PaulStoffregen