PaulStoffregen/Audio #148: GUI: tooltips with images broken

PaulStoffregen, 2016-01-11:
Somewhere we lost tooltips completely.  Then they were fixed, but the ones with images aren't working.

Here's what we're supposed to see.

[image]

Here's what we have now:

[image]

Not a super urgent bug since the right-side documentation is correct... but something to fix eventually.

mamuesp, 2016-01-11:
Well, as I started debugging, the tooltips didn't work at all. I found a ToDo remark, that always the first <p> tag should be used as tooltip. I implemented that, and it worked fine on almost the whole node collection. But some nodes in the HTML help data aren't formatted in a consistent way, so the <p> tags had been wrapped in <div> tags and so on. This I could fix with a little change in the jQuery code, but if there are other <div> in the <p> tags, they are not included by jQuery, because they are not valid HTML. See this explanation: http://stackoverflow.com/questions/17021261/div-are-not-showing-inside-p-tags. So I think we should declare one valid format, evrything else is pilot's error.

closed 2016-01-11 by PaulStoffregen

mamuesp, 2016-01-11:
Fast decision, pragmatic solution ... ;-)

PaulStoffregen, 2016-01-11:
Turns out the old way that "worked" just grabbed the 3rd tag, which was a <p> in some and a <div> in others.  The TODO comment was old, from before some where converted to <div> so an image or more than paragraph could be used.

Eventually I want to create nice images for objects.....

mamuesp, 2016-01-11:
Well, in this case ... always learning, what to keep and what not to keep in the code ... :-)

commit 2f5d57d5 (2016-01-11): GUI: use div for tooltip info, fixes #148