PaulStoffregen/Audio #149: Travis CI check randomly fails

PaulStoffregen, 2016-01-11:
Occasionally the checks are failing, when the code is perfectly fine.

[image]

PaulStoffregen, 2016-01-11:
@ivankravets Any ideas?

ivankravets, 2016-01-12:
PlatformIO 2.7 doesn't depend on any software in host OS. `scons` python package was moved to PlatformIO storage. Looks like Travis CI sometime uses obsolete cache. Please delete it  https://docs.travis-ci.com/user/caching/#Clearing-Caches

PaulStoffregen, 2016-01-12:
These problems really make me wonder if this CI thing is worthwhile.

Yeah, it's nice that Platform IO can do this build.  But we're not really checking if the code actually works, only that all the examples compile.  In the entire history of this library, I do not believe we've ever had a problem where the code didn't compile.  Bugs where it did wrong things have happened, but not simple compile fails.

But in just the last month, this Travis CI build has falsely warned of problems at least 3 times.  That seems pretty unreliable.

ivankravets, 2016-01-12:
@PaulStoffregen this is OK for me if you will disable Travis CI.

>But we're not really checking if the code actually works

I'm working on PlatformIO 3.0 where is planned real hardware embedded unit testing 
https://github.com/platformio/platformio/issues/408