PaulStoffregen/Audio #152: assembler for copy_to_fft_buffer in fft

FrankBoesing, 2016-01-15:
Hi Paul, 

copy_to_fft_buffer could be speed up by a factor of two with the following change:

memcpy_tointerleaveL((int16_t*)destination,(int16_t*)source); //First half
memcpy_tointerleaveL((int16_t*)destination + AUDIO_BLOCK_SAMPLES,(int16_t*)source + AUDIO_BLOCK_SAMPLES/2); //Second half

(two calls of memcpy_tointerleaveL)

Some additional cycles could be saved with a specialized routine.

I want to do a pull-request
Do you want a specialzed routine with more speed or the solution above with two calls which re-uses the existing code ?

FrankBoesing, 2016-01-15:
p.s. it's two times faster, even with the two calls.

FrankBoesing, 2016-01-15:
i tested both variants:
v1 (two calls) is 463 us (for 100 loops)
v2 (specialized) is 432 us (for 100 loops)
the difference is not very dramatic, i'd vote for "v1"

PaulStoffregen, 2016-01-16:
Before optimizing memory copies, I want to explore replacing the arm math library with kcp's optimization.

https://forum.pjrc.com/threads/27905-1024-point-FFT-30-more-efficient-%28experimental%29

FrankBoesing, 2016-01-16:
Ok,
just for the records, this would be the new code:

#include "memcpy_audio.h"

static void copy_to_fft_buffer(void *destination, const void *source)
{
     #if defined(__MK20DX128__) || defined(__MK20DX256__) || 
defined(__MK66FX1M0__)
     //160115 Frank Bösing use assembler.
     //Could be some cylcles faster with a specialized routine
memcpy_tointerleaveL((int16_t*)destination,(int16_t*)source); //First half
     memcpy_tointerleaveL((int16_t*)destination + 
AUDIO_BLOCK_SAMPLES,(int16_t*)source + AUDIO_BLOCK_SAMPLES/2); //Second half
     #else
   // 140312 - PAH - slightly faster copy
     const uint16_t *src = (const uint16_t *)source;
     uint32_t *dst = (uint32_t *)destination;

     for (int i=0; i < AUDIO_BLOCK_SAMPLES; i++) {
         *dst++ = *src++;  // real sample plus a zero for imaginary
     }
     #endif
}

Regards,
Frank.

closed 2017-09-14 by FrankBoesing