PaulStoffregen/Audio #171: biquad filter clicks when adjusted.

myklemykle, 2016-02-29:
I wrote code to allow live MIDI knob-twiddling of the biquad filter, using successive calls to the set methods (setLowpass, setHighpass, etc) while adjusting values for frequency & Q.  I noticed that the filter emits a pronounced click noise with every parameter change, and the noise doesn't seem to get quieter even when the changes in value are relatively microscopic.

I did a little bit more experimentation & determined that even if I call any of those methods repeatedly with the same set of parameters, I still get a click noise every time I make the call.

PaulStoffregen, 2016-02-29:
Maybe try editing the last few lines in setCoefficients(), here:

https://github.com/PaulStoffregen/Audio/blob/master/filter_biquad.cpp#L94

Trouble is, if the filter state isn't reset to a safe value, a filter with resonance that might be stable under all inputs could possible oscillate.  Maybe?  Maybe not?

myklemykle, 2016-02-29:
Thanks, I'll try that.  The docs say there's already the possibility of instability under certain settings ... but even analog filters have their oscillations & such problems.  Dunno what the general solution would be for that, other than some kind of limiting.

myklemykle, 2016-02-29:
Commenting out all three lines does remove the click.  Thanks!

So far I'm not sure if i've hit anything 'unstable'.  If you can suggest values to try, I'll test with them.  I have noticed that there are artifacts when the filter frequency gets close to exactly the input frequency of the input square wave ... perhaps that's what you mean?

I did notice that sweeping either the frequency or the Q through zero leads to clicks.  Is that a bug?  I'm not sure what the meaning of negative Q or negative frequency would be.  But it's interesting that I can pass them, as all the math in the algorithm is signed. The result is some kind of ring-modulated glitchy thing that's probably erroneous, but still sounds kind of cool ...

PaulStoffregen, 2016-03-01:
You know biquads aren't good for low frequency filters, right?  Isn't that mentioned in the design tool docs?

PaulStoffregen, 2016-12-10:
fixed by https://github.com/PaulStoffregen/Audio/commit/e3353358b6f01d40c1cc6b9c3c4bf8a40df84210

closed 2016-12-10 by PaulStoffregen