PaulStoffregen/Audio #172: Audio stops at random times

tobiasebsen, 2016-03-08:
I have programmed a drum-sequencer that plays back wav files samples from an SD card. This surely does a lot of opening and reading files, which actually works very well – no noticeable latency. But...

At random times, after running for anything from a few seconds to a few minutes, it stops.

I have tracked the issue down to being problem with the AudioStream interrupt. Although my AudioOutputI2S object keeps calling `AudioStream::update_all()`, the `AudioStream::isr()` stops being called after a random period of running.

I don't know much about interrupts, so can anyone please shed some light on this?

PaulStoffregen, 2016-03-08:
Are you using the default SD library, or the optimized one from SD_t3.h?

tobiasebsen, 2016-03-08:
The default one. Could that be the source of my problems? And how?

PaulStoffregen, 2016-03-08:
Very likely.  Find and edit SD_t3.h.  See the comments in that file.

Let me know if the optimized SD fixes everything?  (so we know it's the SD lib and not Audio)

If the trouble is within Audio, going to need you to post code and file and instructions to reproduce the problem.

tobiasebsen, 2016-03-08:
Thanks. I will try that and get back to you.

tobiasebsen, 2016-03-08:
It's been running for an hour now without stopping. So it appears to be the SD card that stops the interrupt, although I cannot figure out why.

PaulStoffregen, 2016-03-08:
Oh, I'm pretty sure I know why.  There's a reason I started that complete rewrite of the SD library.  ;)

tobiasebsen, 2016-03-08:
Cool. Feel free to close this issue then.
Any plans for when to enable the Teensy optimized code by default?
Does it have any stability issues, since it's not default?

PaulStoffregen, 2016-06-04:
> Any plans for when to enable the Teensy optimized code by default?
> Does it have any stability issues, since it's not default?

It's very stable, but lacks any ability to write to the card.  Obviously it can't become the default until it's a full functional replacement, able to create, write, append files.

Contributions welcome....

MacroMachines, 2016-08-01:
Would it be possible to supply your partial rewrite as an optional patch for the read sd functions in the default library

PaulStoffregen, 2016-08-01:
All the code is here and open source, so feel free to help yourself to the code...

https://github.com/PaulStoffregen/SD