PaulStoffregen/Audio #185: Distorted audio when playing through USB

dturner, 2016-06-10:
When using the teensy+audio shield I get distorted audio after a few minutes of playing a bass heavy track. Once the distortion starts it gets progressively worse until no audio can be heard. 

Steps to reproduce
- Set the Teensy to USB type Audio
- Load the PassThroughUSB example
- Play a track with a lot of bass. Personally I use "DJs take control" by SL2. 
- Listen to the track

Expected results
- The track plays with no distortion

Actual results
- After a few minutes distortion occurs and gets progressively worse until only glitches are heard

Test environment/equipment
- Teensy 3.1 
- Audio Shield Version 2
- Mac OSX 10.11.5
- Teensy Audio library: latest from github

FrankBoesing, 2016-06-10:
Are you using line out or headphones ?
If headphones: Did you try other headphones ? 
Touch the chip (after you get the distortions): Is it warm ?

dturner, 2016-06-10:
I'm using headphones. I tried with 2 pairs (Sennheiser HD25s and Beyerdynamic DT770 Pros). 

I connected a piezo buzzer to the Teensy DAC pin and (although it was quiet) I could hear the same distortion. I also soldered up a completely new teensy+audio shield and it exhibits the same issue. 

This indicates that the problem is with the Teensy rather than the audio shield. Any advice on how I can debug it further?

dturner, 2016-06-10:
Oh and the teensy chip is slightly warm but I wouldn't say hot. I don't notice any temperature difference from normal usage.

dturner, 2016-06-10:
Bit of further info, the distortion always seems to happen around the 2m 10s mark. Tried on 2 different songs, same result.

FrankBoesing, 2016-06-10:
Please post your question on https://forum.pjrc.com
Read the forum-rules before, and post your sketch if possible.
This is not the place for such questions!

dturner, 2016-06-10:
Sorry, I figured this was the best place to discuss this specific issue. I actually believe this issue is now associated with the Teensy core library so I will file an issue there and link to it from here.

closed 2016-06-11 by dturner

dturner, 2016-06-11:
@FrankBoesing As per your advice I have made a forum post here: https://forum.pjrc.com/threads/34855-Distorted-audio-when-using-USB-input-on-Teensy-3-1?p=106490