PaulStoffregen/Audio #187: Bug in Audio/play_sd_wav.cpp - some WAV files play with stereo left/right channels reversed

el-supremo, 2016-06-27:
There is a bug in the way in which Audio/play_sd_wav.cpp handles some stereo WAV files. The symptom is that the left and right channels are swapped during playback. The demo file which demonstrates this clearly is SDTEST3.WAV. If you listen to it on a PC, at 9 seconds into the recording the guitar is obviously in the left channel. When the file is played back on a Teensy, the guitar is in the right channel.
The reason is that in SDTEST 1, 2 and 3 (but not in SDTEST4.WAV) there is a LIST chunk which is not a multiple of 4 bytes long. When the parser gets to the DATA chunk, the samples which remain in the current block are sent correctly (except for the last one) because the pointer within the first block is correctly offset. But every block after that is handled as though the first sample in the block is the left channel when it is actually the right channel. A block normally would start with samples in order L1,R1,L2,R2 etc. so that sample pairs are (L1,R1), (L2,R2)... But in this case they are, in effect, R0,L1,R1,L2,R2 where R0 is the right channel which belongs with the last sample in the previous block. The sample pairs are now (R0,L1), (R1,L2) etc thereby switching the left and right channels and getting the pairs mixed up.
For some music switching left and right channels is not a problem, but when listening to orchestral music, as an example, it is quite evident when the channels are switched because the violins should be in the left channel.
I don't know what other effect the mixed up pairing of (R0,L1),(R1,L2) will have but it certainly isn't correct.
SDTEST4.WAV also has a LIST chunk but it is a multiple of 4 bytes, so the samples start on a proper 4 byte boundary and this file will play correctly.

I haven't figured out yet how to fix this - still pondering it.

Pete

el-supremo, 2016-06-27:
I've found the problem and issued a pull request but I'm not sure the PR worked.

PaulStoffregen, 2016-07-04:
Thanks for #188

closed 2016-07-04 by PaulStoffregen