PaulStoffregen/Audio #209: The SGTL5000 AutoVolumeControl function does not work

bmillier, 2016-11-09:
The SGTL5000 AutoVolumeControl function does not work as described. There are 3 incorrect lines in the code which result in the Threshold (16-bit), Attack(12-bit) and Decay(12-bit) registers of the SCTL5000 being loaded incorrectly.

In the file "control_sgtl5000.cpp" the following 3 lines

Line#
844 uint8_t thresh = (pow(10, threshold / 20)*0.636)*pow(2, 15);
845 uint8_t att=(1-pow(10,-(attack/(20*44100))))*pow(2,19);
846 uint8_t dec=(1-pow(10,-(decay/(20*44100))))*pow(2,23);

should be replaced with
uint16_t thresh = (pow(10, threshold / 20)*0.636)*pow(2, 15);
uint16_t att=(1-pow(10,-(attack/(20*44100))))*pow(2,19);
uint16_t dec=(1-pow(10,-(decay/(20*44100))))*pow(2,23);

This loads the registers with the proper 16, 12-bit values, and when I tested it out on the Audioshield, the AVC now works properly.

Pensive, 2016-11-09:
That explains the funny results I got when I was playing with it. It also explains the crashing I experienced with certain values. Well done sir! Hope to test it soon...

CyborgEars, 2016-11-11:
I have reported these faulty lines in 2015: https://forum.pjrc.com/threads/29072-Audio-library-control_SGTL5000-cpp-documentation-improvements?p=77939&viewfull=1#post77939

For an in-depth explanation of this issue, read the specification of table 57 to 59 in the SGTL5000 datasheet (rev.6) or my forum entry linked above.

A few days ago, I noticed that, although quite straightforward, the relevant corrections haven't made it into the mainline Audio repository, which is why I've prepared the following commit 9c7671763899a7dcf633f35b847af9b698cc4db8
I did not yet have the time for a thorough review of the possible side effects of this fix on the current code, but as @bmiller has tested this, I have created pull request #210 solving this issue.

@bmillier: did you stumble over my recent commit / the forum post, or did you find this flaw independently?

bmillier, 2016-11-11:
@CyborgEars    No, I solved it myself- the hard way by examining the code, just like you must have done. I had only been using Teensy for a few weeks at that  time, and hadn't even started to follow the forum or github. I follow the forum now though.