PaulStoffregen/Audio #235: SD/XPT2046 IRQ Conflict?

jshoek, 2017-05-22:
If using Paul's FFT code for playing audio from the SD audio shield board while detecting a touch from the ILI9341/XPT2046 display, is there a possible IRQ conflict sinse both are on the SPI bus? I know that Paul suggested enabling/disabling the IRQ from the XPT2046 to manage when IRQ are allowed. Does the SD audio shield board also use SPI IRQ's? I could see a conflict between IRQ's if both the SD and XPT2046 are using the same interrupt pin, but I don't see any such #define assignment in the code for the SD. I have the audio shield soldered directly to the Teensy 3.2 (stack-up). Since all 28 pins are connected together, is this a potential problem?

What I am experiencing is the (ts.touched) is constantly being triggered even though the screen is not being touched while the FFT code is being executed. This is shown on the USB serial monitor as X and Y touch values of 0 to 2 (obviously not valid). The FFT is functioning correctly by displaying volume and frequency measurements of the audio on the ILI9341. Are there any reports of this same problem from others?

jshoek, 2017-05-22:
Should I try turning off XPT2046 touch IRQ's when executing FFT? I just think it is odd that other opporating modes that do not use FFT don't have this problem.

PaulStoffregen, 2017-05-23:
I see you closed this issue.  Is the problem resolved?

To answer your original question, no, no other reports of this problem so far.  You're the first.

If you reopen this issue, please post a complete program which demonstrates the problem.

jshoek, 2017-05-23:
Paul, I did not mean to close the issue. I did add a check in the Touch_Mode function to determine if the X coordinate was below a minimal value and it seems to prevent these false touch values from being processed, but they are still being detected:

void Touch_Mode(){ //Mode and color palette selection
  if(ts.touched()) {
      TS_Point p = ts.getPoint(); //Get touch points
      if(p.x <= 5) { //Check x coordinate for false touch
        cordx = 0; //Clear touch coordinates
        cordy = 0;
      }
      else {
      Serial.println();
      Serial.print("Pressure = "); Serial.print(p.z); //Print position, pressure
      Serial.print(", x = "); Serial.print(p.x);
      cordx = (p.x);
      Serial.print(", y = "); Serial.println(p.y);
      cordy = (p.y);
      delay(100); //Is this enough delay or to much?        
      }
  }
How ever I still wonder why this is happening only when the FFT is being processed. I probably never noticed this before as I was not looking for it. The entire code is quite extensive so here it is. There are three modes that use the FFT; Audio, Sequence and Converge. In each of these I convert the FFT float to an integer value for displaying. Just note that I am a novice C++ coder, not a Pro like you. 

[Final_Version_with_color_no_XBee3-170523a.zip](https://github.com/PaulStoffregen/Audio/files/1023625/Final_Version_with_color_no_XBee3-170523a.zip)

If you see anything obvious let me know. Thanks much.