PaulStoffregen/Audio #239: 3.5 dacs memory usage

BleepLabs, 2017-06-11:
This example returns some odd results for memory useage on a 3.5 

`#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>

AudioSynthWaveformSineModulated      waveform1;      //xy=224,285
AudioSynthWaveformSineModulated      waveform2;      //xy=224,285
AudioOutputAnalogStereo  dacs1;          //xy=538,351
AudioConnection          patchCord1(waveform1, 0, dacs1, 0); //remove this connection and memory is still off
AudioConnection          patchCord2(waveform2, 0, dacs1, 1);  //remove this connection and memory results seem normal

uint32_t pm1;

void setup() {
  
  AudioNoInterrupts();
  //dacs1.analogReference(INTERNAL); //int and external have same effect
  AudioMemory(25);
  Serial.begin(9600);

  waveform1.amplitude(1);
  waveform1.frequency(220);

  waveform2.amplitude(1);
  waveform2.frequency(110);

  AudioInterrupts();
}

void loop()
{
  
  if (millis() - pm1 > 200) {
    pm1 = millis();

    Serial.print(" proc:");
    Serial.print(AudioProcessorUsageMax());
    Serial.print("  mem:");
    Serial.println(AudioMemoryUsageMax());
    Serial.println();

    AudioProcessorUsageMaxReset();
    AudioMemoryUsageMaxReset();
  }

} 
`
Removing the connection to dacs1 and it reports more reasonable numbers.
I have tried this with various objects connected to the dacs. 
I don't have a 3.6 to test on right now.

FrankBoesing, 2018-09-25:
This sketch works smoothly. Probably the problem was solved in the past.
I think the issue can be closed.

PaulStoffregen, 2018-09-25:
Oh, yes, early versions AudioOutputAnalogStereo had a couple nasty bugs (entirely my fault).  It was fixed long ago.

closed 2018-09-25 by PaulStoffregen