PaulStoffregen/Audio #245: return block count

WMXZ-EU, 2017-08-23:
suggest that available() return block count instead of Boolean "count>0"
This allows a better control on how many blocks are used for rms
e.g. by using something like
while((nblk=rms.available())<5) yield();

PaulStoffregen, 2017-08-23:
As a general design goal, I've been trying to keep the audio API to Arduino sketches using milliseconds for time and -1.0 to +1.0 for signal levels.  Some exceptions (mistakes) have been made along the way.  But the 2 main goals for using those units are Arduino-style simplicity (even if not the most optimal code) and some degree of hardware independence to hopefully scale up to more powerful hardware in coming years.

WMXZ-EU, 2017-08-23:
Cannot relate your comment to my suggestion.

You are talking about times (millisecond).

I was suggesting to replace
bool available(void) { return count>0;}
by
uint32_t available(void) {return count;}