PaulStoffregen/Audio #252: [Feature] Allow wiring of low frequency signals

dronus, 2018-02-15:
It would be cool if there would be an equivalent to AudioConnection for other control inputs, that only exhibit by setters like `.frequency()` or `.amplitude()` now.

This way we could implement for example MIDI input nodes exposing MIDI notes as envelope control signals, like usual patchable synthesizers would use ('control voltage' or CV patching).

Also MIDI control values could easily be mapped. 

Finnaly the Audio System Designer could merge into node-red then, using the already available node-red nodes for value mapping.

Currently additional code is needed to implement even simple instruments, as the Audio System Designer does not create setter code.

42loop, 2018-02-16:
something like this : https://github.com/EtienneGameSeed/node-blue could be woven together with pauls audio library - which is definately a lot of work. I tried some time ago wether these can work together in a single sketch by copying from the audio design tool and the node-blue tool, which works fine - as far as i tested. node-red on the other hand seems a bit farther away, since it is focussed very much on network data streams.

42loop, 2018-02-17:
a thread on the forums can be found here:
https://forum.pjrc.com/threads/42282-Node-Red-Audio-System-Design-Tool-Modification

PaulStoffregen, 2018-02-19:
Interesting ideas.  Just so you know, I personally can not put time into this right now.