PaulStoffregen/Audio #264: Online Audio Sys Des Tool doc update for getBuffer(); add available();

Defragster, 2018-09-26:
### Description

Error in and Undocumented functions on https://www.pjrc.com/teensy/gui/?info=AudioPlayQueue
AudioPlayQueue details **getBuffer();** suggesting it will return NULL, but it will not.  Also the **available();** function is not documented

### Steps To Reproduce Problem

See https://forum.pjrc.com/threads/53860-Audio-play_qeue()-documentation 

Not seeing where the source lives - this is a suggested text change spotted by Frank_B with his review of forum posted text.

FrankBoesing, 2018-10-02:
https://github.com/PaulStoffregen/Audio/pull/268/commits/2254de01a97cca187d647d3527799209318cef75

Defragster, 2018-10-02:
Nice Frank - I didn't take the next step to locate the source lines.

https://github.com/PaulStoffregen/Audio/blob/2254de01a97cca187d647d3527799209318cef75/gui/index.html#L1757