PaulStoffregen/Audio #302: BUG in the WavFilePlayer example

mariocaptain, 2019-06-07:
### Description
The WavFilePlayer example doesn't work with Teensy 3.5's BUILTIN_SDCARD pin. 

SD.begin(BUILTIN_SDCARD) freezes as long as there is any AudioPlaySdWav.play() call in the code.

I have tested with many brand new Teensy 3.5 boards and SD cards.

Also I have tested and can confirm that this problem doesn't happen when an external SPI SD module. In that case, in the below code I just replace BUILTIN_SDCARD with the CS pin connected to the module (8 in my case). So the same code works with an external SD module.

### Steps To Reproduce Problem
You can use my code below, which is essentially the stock WavFilePlayer example.

### Hardware & Software

Board: Teensy 3.5
Shields / modules used: none
Arduino IDE version: 1.8.9
Teensyduino version: 1.46
Operating system & version: Windows 10

### Arduino Sketch

```cpp
[/C0DE]
Reply With Quote Reply With Quote Today, 09:11 AM #6
mariocaptain  mariocaptain is offline
Member
Join Date
Mar 2017
Posts
21
This is my code:
Code:
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>

// GUItool: begin automatically generated code
AudioPlaySdWav           playSdWav1;     //xy=343.00000381469727,3690.2000064849854
AudioOutputAnalogStereo  dacs1;          //xy=525.0000076293945,3693.2000064849854
AudioConnection          patchCord1(playSdWav1, 0, dacs1, 0);
AudioConnection          patchCord2(playSdWav1, 1, dacs1, 1);
// GUItool: end automatically generated code

// Use these with the Teensy 3.5 & 3.6 SD card
#define SDCARD_CS_PIN    BUILTIN_SDCARD
#define SDCARD_MOSI_PIN  11  // not actually used
#define SDCARD_SCK_PIN   13  // not actually used

void setup() {
  Serial.begin(115200);
  AudioMemory(8);
  SPI.setMOSI(SDCARD_MOSI_PIN);
  SPI.setSCK(SDCARD_SCK_PIN);
  if (!(SD.begin(SDCARD_CS_PIN))) {
    while (1) {
      Serial.println("Unable to access the SD card");
      delay(500);
    }
  }
}

void playFile(const char *filename)
{
  Serial.print("Playing file: ");
  Serial.println(filename);

  playSdWav1.play(filename);
  delay(5);

  // Simply wait for the file to finish playing.
  while (playSdWav1.isPlaying()) {
  }
}

void loop() {
  playFile("SDTEST1.WAV");  // filenames are always uppercase 8.3 format
  delay(3000);
}

eaconner, 2019-06-10:
I'm having the same issue. Switching to a previous version of Arduino and Teensyduino works, seems like the latest version broke something with ```AudioPlaySdWav```.

eaconner, 2019-06-10:
Looks like PR #294 should fix the issue but I haven't tested it yet.

manitou48, 2019-06-10:
The same hang occurs at SD.open() in  play_sd_raw.cpp.  The PR #294 does not consider problem in  	play_sd_raw.cpp

tsandmann, 2019-06-10:
I have updated PR #294 to cover `AudioPlaySdRaw` as well.