PaulStoffregen/Audio #312: AudioSynthWaveform::begin doesn't set the phase to 0

skybrian, 2020-01-14:
Calling AudioSynthWaveform::begin() sets phase_offset to zero, but it doesn't clear the phase_accumulator field, so the wave doesn't actually start from zero.

Similarly for phase(), although in that case, maybe you just want to shift the phase of the running waveform a bit, instead of restarting from the beginning? If so, maybe the documentation should be clarified.

I found this when playing two sawtooth waves that are slightly detuned; the note starts at a different place each time. (Sometimes that's what you want, but not always.)

skybrian, 2020-01-14:
Here is how I edited my local copy. (I also added the code disabling interrupts.)

```
	void begin(short t_type) {
		__disable_irq();
		phase_offset = 0;
		phase_accumulator = 0;
		tone_type = t_type;
		__enable_irq();
	}
	void begin(float t_amp, float t_freq, short t_type) {
		__disable_irq();
		amplitude(t_amp);
		frequency(t_freq);
		phase_offset = 0;
		phase_accumulator = 0;
		tone_type = t_type;
		__enable_irq();
	}
```

ErikDorstel, 2020-01-14:
I also came across this problem some time ago. Since I also have the requirement to set the phase to a certain value during runtime, I came up with another solution.

https://github.com/PaulStoffregen/Audio/pull/275

skybrian, 2020-01-14:
If `begin()` cleared the phase accumulator, you could call `phase()` immediately after to set the phase. That seems a bit obscure though, compared to `phase()` actually setting the phase.

Advancing the phase a bit would be more clearly written as something like: `wave.phase(getPhase() + delta)`;