PaulStoffregen/Audio #324: SGTL5000 I2C not working while config_i2s has not been called

siteswapjuggler, 2020-03-25:
### Description

I'm trying to get I2C communication working with a SGTL5000 on an Audio board, I don't need the other objects of the library for now just a proof of communication for later use with I2S with and a Raspberry Pi.

Well it came out that the SGTL I2C communication isn't responding while I do not declare an AudioOutputI2S before the enable() call. I don't succeed figuring out why (by extension the scanner exemple in Teensy wire library doesn't succed in SGTL5000 detection either).

### Steps To Reproduce Problem

In control_sgt5000.cpp uncomment line 517-520 to print the CHIP_ID. 
Use the code above. CHIP_ID stay at 0 instead of giving us a valid result

### Hardware & Software

Teensy 3.1
Audio Shield Rev B
Arduino IDE version 1.8.12
Teensyduino version 1.51
Operating system & version Mac OS 10.13.6

### Arduino Sketch

```cpp
#include <Audio.h>
#include <Wire.h>

//AudioOutputI2S       audioOutput; // uncomment this line to get the appropriate result
AudioControlSGTL5000 audioShield;

void setup(void)
{
  Serial.begin(9600);
  while (!Serial) ;
  delay(3000);
  audioShield.enable();
  Serial.println("setup done");
}

void loop(void)
{
}
```

FrankBoesing, 2020-03-25:
The SGTL needs a clock - which is not present without I2S.
Without clock, it can not respond to the I2C communication.
Please read the datasheet.

PaulStoffregen, 2020-03-25:
Yup, the chip stays in a low power reset state until it hears 8 pulses on MCLK.

siteswapjuggler, 2020-03-25:
Ok ok sorry for my misunderstanding just found it page 9 and 10 of the datasheet. Thanks for your anwsers.

closed 2020-03-25 by siteswapjuggler