PaulStoffregen/Audio #328: Random noise when playing WAV files

woodencase01, 2020-03-28:
### Description

The WAV player is emmiting random high pitch noise.
Nothing from the original WAV can be recognized.

Used to work when I started working on this project, about 3 month ago.

### Steps To Reproduce Problem

Upload the example Sketch WavFilePlayer.ino

### Hardware & Software

Teensy 3.2
Audio board REV B
VisualStudio Code / PlatformIO
Teensy 4.8.0:
PACKAGES:
 - framework-arduinoteensy 1.151.0 (1.51)
 - tool-teensy 1.151.200304 (1.51) 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
Windows 10

### Arduino Sketch

```cpp
void loop() {
  playFile("SDTEST1.WAV");  // filenames are always uppercase 8.3 format
  delay(500);
  playFile("SDTEST2.WAV");
  delay(500);
  playFile("SDTEST3.WAV");
  delay(500);
  playFile("SDTEST4.WAV");
  delay(1500);
}
```

### Errors or Incorrect Output

High pitch noise.

woodencase01, 2020-03-28:
Ran some tests with older versions:
Tested with teensy@4.5.0 - teensy@4.3.0
Using older audio commits have errors with "'arm_dcache_flush_delete'"

`
.pio\libdeps\older\Audio@src-417c8d857b777deee43c3e9774edb685\output_i2s.cpp: In static member function 'static void AudioOutputI2S::isr()':
.pio\libdeps\older\Audio@src-417c8d857b777deee43c3e9774edb685\output_i2s.cpp:137:58: error: 'arm_dcache_flush_delete' was not declared in this scope
Compiling .pio\build\older\lib4a7\Audio@src-417c8d857b777deee43c3e9774edb685\play_sd_wav.cpp.o
  arm_dcache_flush_delete(dest, sizeof(i2s_tx_buffer) / 2 );
`

FrankBoesing, 2020-03-28:
that's usually poor soldering or no soldering at all, or general poor connections (china patch wires)

woodencase01, 2020-03-29:
> that's usually poor soldering or no soldering at all, or general poor connections (china patch wires)

Thanks. I indeed has it set on a breadboard with wires...

I'll give it a try as soon as I can solder it.

PaulStoffregen, 2020-04-05:
Issues like "random high pitch noise" are often caused by ground loops.  The simplest way to check is listening with headphones, and make sure Teensy has no other connections to equipment other than your computer.

PaulStoffregen, 2020-04-05:
Closing this issue, since it's almost certainly not a software defect.

closed 2020-04-05 by PaulStoffregen

woodencase01, 2020-04-05:
You guy were right.
I desoldered the pins I had on my teensy and installed these:
SSQ-114-03-T-S
[image]
Works like a charm!

Thanks for your kind feedback.