PaulStoffregen/Audio #337: SleepyDog with WavFilePlayer Example Not Working

Casey10110, 2020-04-24:
I'm trying to figure out why Adafruit's branch of this library is failing when using the SleepyDog library with a Feather M4 Express (SAMD51). When I replace the delays in the WavFilePlayer example code with Watchdog.sleep(500), it crashes after a few cycles and ceases to sleep. Any idea what is preventing it from sleeping? It's like something is triggering the _WFI() before the WDT is able to ...

`
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>
#include <Adafruit_SleepyDog.h>

AudioPlaySdWav           playWav1;
// Use one of these 3 output types: Digital I2S, Digital S/PDIF, or Analog DAC
AudioOutputI2S           audioOutput;
AudioConnection          patchCord1(playWav1, 0, audioOutput, 0);
AudioConnection          patchCord2(playWav1, 1, audioOutput, 1);
AudioControlSGTL5000     sgtl5000_1;

// Use these with the Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14

// Use these with the Teensy 3.5 & 3.6 SD card
//#define SDCARD_CS_PIN    BUILTIN_SDCARD
//#define SDCARD_MOSI_PIN  11  // not actually used
//#define SDCARD_SCK_PIN   13  // not actually used

// Use these for the SD+Wiz820 or other adaptors
//#define SDCARD_CS_PIN    4
//#define SDCARD_MOSI_PIN  11
//#define SDCARD_SCK_PIN   13

void setup() {
  Serial.begin(9600);

  // Audio connections require memory to work.  For more
  // detailed information, see the MemoryAndCpuUsage example
  AudioMemory(8);

  // Comment these out if not using the audio adaptor board.
  // This may wait forever if the SDA & SCL pins lack
  // pullup resistors
  sgtl5000_1.enable();
  sgtl5000_1.volume(0.5);

  SPI.setMOSI(SDCARD_MOSI_PIN);
  SPI.setSCK(SDCARD_SCK_PIN);
  if (!(SD.begin(SDCARD_CS_PIN))) {
    // stop here, but print a message repetitively
    while (1) {
      Serial.println("Unable to access the SD card");
      delay(500);
    }
  }
}

void playFile(const char *filename)
{
  Serial.print("Playing file: ");
  Serial.println(filename);

  // Start playing the file.  This sketch continues to
  // run while the file plays.
  playWav1.play(filename);

  // A brief delay for the library read WAV info
  delay(5);

  // Simply wait for the file to finish playing.
  while (playWav1.isPlaying()) {
    // uncomment these lines if you audio shield
    // has the optional volume pot soldered
    //float vol = analogRead(15);
    //vol = vol / 1024;
    // sgtl5000_1.volume(vol);
  }
}

void loop() {
  playFile("SDTEST1.WAV");  // filenames are always uppercase 8.3 format
  Watchdog.sleep(500);
  playFile("SDTEST2.WAV");
  Watchdog.sleep(500);
  playFile("SDTEST3.WAV");
  Watchdog.sleep(500);
  playFile("SDTEST4.WAV");
  Watchdog.sleep(500);
}`

Casey10110, 2020-04-25:
The issue does not appear to have to do with Watchdog, as now I am getting crashes using delay() as well ...

Casey10110, 2020-04-25:
Cancel that, I actually had a sleep in there next to the delay that I oversaw, and that is causing the crash. I am also now using:

`AudioInterrupts(); Watchdog.sleep(500); NoAudioInterrupts();`

and this doesn't seem to help.

Casey10110, 2020-04-25:
Cancel that, I actually had a sleep in there next to the delay that I oversaw, and that is causing the crash. I am also now using:

AudioInterrupts(); Watchdog.sleep(500); NoAudioInterrupts();

and this doesn't seem to help.

FrankBoesing, 2020-04-25:
Please ask Adafruit. Thisvariant of the library is not written for that CPU.

Casey10110, 2020-04-25:
OK, I did but haven't received any response. Thank you, Frank!

FrankBoesing, 2020-04-25:
The library needs interrupts - it's likely they wake the CPU.

Casey10110, 2020-04-25:
For sure Frank, I've tried to get rid of them all but not yet to any avail. NVIC_DisableIRQ(DMAC_0_IRQn); crashes the sleep function entirely. I've reposted in the Adafruit forum we'll see what they say ...

Casey10110, 2020-04-25:
To update, it appears this works: 

`// Minimal Crashing Example
// Sleep crashes after a certain (seemingly random) number of cycles

#include <Adafruit_SleepyDog.h>
#include <play_fs_wav.h>
#include <Audio.h>

AudioPlayFSWav           playFSWav;
AudioOutputI2S           audioOutput;
AudioConnection          patchCord1(playFSWav, 0, audioOutput, 1);
AudioConnection          patchCord2(playFSWav, 1, audioOutput, 0);

 
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake
  Serial.begin(115200);
  while(!Serial); // wait for Arduino Serial Monitor (native USB boards)
  Serial.println("Crash Demo!");
  Serial.println();
}

//watch the regular pattern change after a certain number of cycles (sleep drops out)
void loop() {
  Serial.println("Going to sleep in one second...");
  AudioNoInterrupts();
  DMAC->Channel[0].CHCTRLA.bit.ENABLE  = 0;
  Watchdog.sleep(64); // Sleep for 64 ms, this drops out after a while
  OSCCTRL->DFLLVAL.reg = OSCCTRL->DFLLVAL.reg;
  while ( OSCCTRL->DFLLSYNC.bit.DFLLVAL );
  DMAC->Channel[0].CHCTRLA.bit.ENABLE  = 1;
  AudioInterrupts();
  digitalWrite(LED_BUILTIN, LOW); // Show we're asleep
  AudioNoInterrupts();
  DMAC->Channel[0].CHCTRLA.bit.ENABLE  = 0;
  Watchdog.sleep(64);  // Sleep for 64 ms, this drops out after a while
  OSCCTRL->DFLLVAL.reg = OSCCTRL->DFLLVAL.reg;
  while ( OSCCTRL->DFLLSYNC.bit.DFLLVAL );
  DMAC->Channel[0].CHCTRLA.bit.ENABLE  = 1;
  AudioInterrupts();
  digitalWrite(LED_BUILTIN, HIGH); // Show we're awake again
}
`

closed 2020-04-25 by Casey10110