PaulStoffregen/Audio #344: Where is the source code for AudioInputUSB object?

hhaudio, 2020-05-21:
Hi, I am working on a project for which an understanding and possible modification of the low level function of the AudioInputUSB object would be very useful. 

I am looking for the source code for this object but it doesn't appear to be included in this repo. It is not among the files labeled input, and a search for 'usb' within the repo returns only example files and some files relation to git management and the design tool ui. 

Where can I find the source code for AudioInputUSB object?

FrankBoesing, 2020-05-21:
https://github.com/PaulStoffregen/cores/blob/master/teensy4/usb_audio.cpp

PaulStoffregen, 2020-05-21:
Or here if using Teensy 3.x: https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_audio.cpp

h4yn0nnym0u5e, 2024-09-14:
Answered, so can be closed

closed 2024-09-17 by hhaudio