PaulStoffregen/Audio #354: Audio library always dependant on builtin SD library

NuclearManD, 2020-06-05:
### Description

I am trying to use the beta SdFat library so I can connect my SD Card to the Teensy 4.1 with the builtin SD card slot.  Uses SDIO - AFAIK SD library does not yet support this.

There are forum posts of similar problems, for example someone trying to use Audio and SdFat together: https://forum.pjrc.com/threads/34327-Audio-library-and-SdFat-Library

There is also an open issue to add SDIO support to SD library: https://github.com/arduino/ArduinoCore-API/issues/9

That would solve my issue but not fix the underlying problem that the Audio library cannot be used alongside SdFat

### Steps To Reproduce Problem

Get SdFat from here: https://github.com/greiman/SdFat-beta
I'm using SdFat-beta so that I can use SDIO with Teensy 4.1 (as stated above)

Try to compile this:

```cpp

#include "SdFat.h"
#include <Audio.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Errors on pastebin (there's a lot of them): https://pastebin.com/h2Ewc45P
I included the entire copy from Arduino in the paste above, scroll to the bottom to just look at error messages.

### Hardware & Software

Board:  Teensy 4.1
Shields / modules used: SD Card inserted, [Sparkfun I2S Audio Breakout](https://www.sparkfun.com/products/14809)
Arduino IDE version 1.8.12
Teensyduino version  1.5.2
Operating system & version  Linux Mint, not sure the version
Any other software or hardware?

### Arduino Sketch

```cpp

#include "SdFat.h"
#include <Audio.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Errors or Incorrect Output

There's a lot of output so I put it in this paste: https://pastebin.com/h2Ewc45P

PaulStoffregen, 2020-06-05:
This is on my high priority list to fix in 1.53.  Or at least partially fix, with further improvements coming in the future.

NuclearManD, 2020-06-05:
I wonder if it could be designed to read the file from any given stream?  That would remove the dependency of the SD library and make it more flexible.  I would be happy to make a fork with that capability.

Of course, I'm pretty sure that would be a breaking change.  I'm new to the Audio library so I'm not real familiar with the API yet - I haven't tried to load anything from an SD card using this library.

Ideally we'd want a solution that:
1. Still only requires the inclusion of Audio.h - does not require additional headers to use some audio streams
2. Removes dependency on the SD library for programs that do not need a filesystem
3. Does not require setting weird macros
4. Sketch does not need some sort of bridge from the filesystem to the Audio library

It would also be nice to be able to load files from other filesystems - SD isn't the only way filesystems are implemented on MCUs, even if they are the most common.

Those are my thoughts, I'll post more if I have any ideas on how to fix it.  It seems tough due to the way all the headers and Arduino is set up though, at least if we want to maintain backwards compatibility.

PaulStoffregen, 2020-06-05:
> if it could be designed to read the file from any given stream?

Yes, that is exactly the plan.  Sorry, I don't have time right now to write a lengthy message.

Stass48, 2023-11-28:
Hello, I have problem like this. Now audio library use sdfat, but I can't play file from sdfat card, because method play(); haven't SdFs argument. For example:

SdFs card1;
...

Mp3player.play(&card1, "test.mp3");

I get many warnings while compiling and the code not working.

How to chose source sd card for playing files if I have two or more sd cards on one teensy?

Thank you.

FrankBoesing, 2023-11-28:
Create a PR for the Codecs lib, I'll merge it.

FrankBoesing, 2023-11-28:
Filesystem: Yes, unfortunately there is still a lot more open, but as with so many projects in teensyland, it was started but never continued. The initial ideas were good, but then work on them was obviously stopped, so a lot of things that would only be possible with minimal changes there can't be implemented. That's a bit sad. 

But as far as audio is concerned, I can recommend a Raspberry (zero 2) as an alternative.

Stass48, 2023-11-28:
The raspberry pi is not suitable due to the need for instant booting after power is applied.

FrankBoesing, 2023-11-28:
Too bad. It boots up under 10 Secs (with some things disabled) and can go to sleep mode.  But if instant booting is mandatory, there is of course little you can do.

h4yn0nnym0u5e, 2023-11-28:
Does [this PR](https://github.com/PaulStoffregen/Audio/pull/451) work for your use case?

Stass48, 2023-11-28:
Hah))) I solved the problem just few minutes ago, from this thread. Something different, from them, but it works! Thank you ❤️