PaulStoffregen/Audio #412: Hard-coded buffer length in play_sd_wav.cpp

StephanLuik, 2021-11-08:
### Description

In the **update()** method of **play_sd_wav.cpp**, the buffer array size is hard-coded to 512 samples. When using a value other than 128 for **AUDIO_BLOCK_SAMPLES**, the buffer length is no longer a multiple and for some files it will introduce audible gaps in the playback. 

### Steps To Reproduce Problem

Change the **AUDIO_BLOCK_SAMPLES** constant to 160 and play back a simple sine wave file.

### Hardware & Software

Teensy 4.1 with pins
Audio Adaptor rev. D
Arduino version 1.8.13
Teensyduino version 1.53
Windows 10 Pro 21H1

### Arduino Sketch

### Errors or Incorrect Output

FrankBoesing, 2021-11-08:
**AUDIO\_BLOCK\_SAMPLES has to be a power of two anyway, for other parts of the library.**

StephanLuik, 2021-11-09:
Still, if you change AUDIO_BLOCK_SAMPLES to a power of two, say 256, then the hard coded array size of 512 will fail also. It is never a good idea to use hard-coded constants in source code.

h4yn0nnym0u5e, 2023-03-19:
I think this is fixed in PR #448