PaulStoffregen/Audio #425: play_sd_wav includes <SD.h>, which breaks Platform IO

eyuan-creare, 2021-12-12:
### Description
play_sd_wav.cpp includes <SD.h>
- SD.h references the typedef: "[DateTimeFields](https://github.com/PaulStoffregen/SD/blob/98106a0aa089283ca780f5921ba294d5d3dd20d6/src/SD.h#L135)"
    - DateTimeFields is found in [core_pins.h](https://github.com/PaulStoffregen/cores/blob/e04c6eace5cf11999481b5fe12ccb454127d3f0a/teensy/core_pins.h#L2548), which I believe is meant for Teensy 2.0.  

When I build in Platform I/O for micromod and include the Teensy Audio repo, a different core_pins.h is used that does not have DateTimeFields is not declared, among other symbols.

### Steps To Reproduce Problem
Build Platform I/O using platformio.init:
```
[env:teensymm]
platform = teensy@4.14.0
framework = arduino
board = teensymm
lib_deps = 
      https://github.com/Tympan/Tympan_Library.git#Tympan_MicroMod
	https://github.com/PaulStoffregen/Audio.git
	https://github.com/PaulStoffregen/Wire.git
	https://github.com/PaulStoffregen/SPI.git
     	https://github.com/PaulStoffregen/SD.git
	https://github.com/PaulStoffregen/SdFat.git
    	https://github.com/PaulStoffregen/Time.git

### Hardware & Software
Board: Micromod with custom board
Shields / modules used:
Arduino IDE version: Platform I/O
Teensyduino version: 1.55
Operating system & version: Win10
Any other software or hardware? Platform I/O

### Arduino Sketch

### Errors or Incorrect Output

PaulStoffregen, 2021-12-17:
You're using the new SD.h with an old copy of the Teensy core library.

The officially supported way to get a complete set of consistent library is with Arduino and the Teensyduino installer.  We're currently at 1.56-beta4 for the latest.

https://forum.pjrc.com/threads/68972-Teensyduino-1-56-Beta-4

PlatformIO is not officially supported.  If you install the correct set of libraries, it probably will work.  But you're on your own to get that right.  Best advice I can offer is to use Arduino + Teensyduino 1.56-beta4 as a guide to get the complete set of consistent libraries.

closed 2021-12-17 by PaulStoffregen

eyuan-creare, 2021-12-17:
@PaulStoffregen thank you for reviewing this, and for the support you provide.