PaulStoffregen/Audio #461: Where is AudioStream.h?

hhaudio, 2023-04-27:
Hi. I am studying the audio library structure for the purpose of writing my own custom objects. They all reference a parent class named AudioStream, and include AudioStream.h but it appears the file AudioStream.h is not actually included in the repo. Maybe I'm missing something, but it's not in the main folder and when I search within the repo it just shows all the times it's included. Where can I find the actual file so I can see the parent class?

dkadish, 2023-09-05:
AudioStream.h is part of the Teensyduino core, so you'll find it in the core for your particular Teensy version (e.g. Teensy 4.x is in https://github.com/PaulStoffregen/cores/blob/master/teensy4/AudioStream.h). It's not part of this repo.

Also, this is probably not the right forum for this kind of question in the future. This is more for reporting issues with the code (rather than asking for help about it). The PJRC forum (https://forum.pjrc.com/forum.php) is probably a better bet, or something like https://stackoverflow.com/

PaulStoffregen, 2023-09-09:
Comments added, hopefully should answer this question for anyone in the future.

https://github.com/PaulStoffregen/Audio/commit/f3777604cfa07f795c5bf6340c4abf967fbf4ffe

closed 2023-09-09 by PaulStoffregen