PaulStoffregen/Audio #479: Error in input_tdm.cpp and input_tdm2.cpp

palmerr23, 2024-09-04:
### Description

When a device that uses 16 bit TDM slots is connected, input data is corrupted on the all odd numbered channels. The issue is not revealed with 32-bit slots as the odd channels are ignored.

This is because of an incorrect calculation on line 102:

` *dest2++ = (in1 << 16) | (in2 & 0x0000FFFF);`

This code swaps the samples, putting the first sample in the high part, and the second sample in the low part of the 32-bit destination word. This is opposite to the behaviour of dest1 on line 101:		

  `*dest1++ = (in1 >> 16) | (in2 & 0xFFFF0000);`

**the solution is to substitute line 102 with this line of code in both files:**

  `*dest2++ = (in1 & 0x0000FFFF)  | (in2 << 16);`

The solution has been tested on CS42448 and TLV320AIC3104 hardware in TDM mode.

### Steps To Reproduce Problem

16-bit TDM hardware is required, such as my new 4xTLV320AIC3104 board. 

Any code that inputs L&R TDM channels will work.

### Hardware & Software

Board 
  T4.0
Shields / modules used New 
  TLV320AIC3104 board
Arduino IDE version 
  2.3.2
Teensyduino version 
  1.59
Operating system & version 
  Win11
Any other software or hardware?

### Arduino Sketch

#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>
#include "control_tlv320aic3104.h"

#define BLOX 40
AudioInputTDM             tdm1;           
AudioOutputUSB           usb1;  
AudioConnection          patchCord15(tdm1, 0, usb1, 0);
AudioConnection          patchCord16(tdm1, 1, usb1, 1);
AudioControlTLV320AIC3104 aic;

void setup() 
{
  AudioMemory(BLOX);
  Wire.begin();

  aic.audioMode(4, true, AICMODE_TDM);
  aic.inputMode(AIC_SINGLE);
  aic.enable( ); 
  aic.gain(0); 
}

void loop() 
{
}

### Errors or Incorrect Output

Right channel input shows a spike at (22.05 -f) kHz, with an amplitude that varies with the input.
[image]
[image]

h4yn0nnym0u5e, 2024-09-04:
This is fixed in #440

palmerr23, 2024-09-04:
Thanks Jonathan for the update - I've also just posted a PR. Hopefully PS will get around to merging one of them in the near future.

See PRs 440 and 480

h4yn0nnym0u5e, 2024-09-04:
Hopefully ... #440 is over 2 years old, but maybe Teensyduino 1.60 will involve some massive Audio library housekeeping

palmerr23, 2024-09-08:
PR 440 has now been merged.

closed 2024-09-08 by palmerr23