PaulStoffregen/Audio #496: Audio connection arrays

xgg69, 2025-02-09:
for simpler polysynth design I want to have the option to connect, for example, an array of oscillators (waveforms) to an array of filters or even a single noise generator to multiple separate mixers in an array

and if this exists already, I'm just too lazy to read the wiki :)

h4yn0nnym0u5e, 2025-02-10:
[This](https://forum.pjrc.com/index.php?threads/audio-system-design-tool-update.65740/) may be some way towards what you want. The author is still active-ish on the forum, though development of this tool seems to have stalled a couple of years back. It just about works OK (I do use it on occasion), but is pretty rough around the edges with lots of experimental and unfinished features, and precious little documentation.

xgg69, 2025-02-10:
sorry, didn't mention this, I was talking about in arduino code instead of having

`AudioConnection       connName(src, dest);`
`AudioConnection       connName(src, nOut, dest, nInput);`

as a single source and single destination but having for example

`AudioConnectionArray        connName(src[], dest[]);`
`AudioConnectionArray        connName(src[], nOut, dest[], nInput);`

multiple sources and destinations as arrays
or even connecting a single output to multiple inputs in an array

`AudioConnectionArray        connName(src, dest[]);`
`AudioConnectionArray        connName(src, nOut, dest[], nInput);`